adamff-dev/ESP8266-Captive-Portal #6: how to change wifi ssid

aniketstark, 2019-10-09:
how to change wifi ssid name

BlueArduino20, 2019-10-09:
There's a very clear line of code on the .ino file that defines the SSID...

``#define SSID_NAME "My SSID"``

aniketstark, 2019-10-09:
when i exporting ino file into bin file it gaves error like that
Arduino: 1.8.10 (Windows XP), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

panic: Failed to find GetFileInformationByHandleEx procedure in kernel32.dll: The specified procedure could not be found.

goroutine 1 [running]:
syscall.(*LazyProc).mustFind(0x11c459e0)
	/home/jenkins/go1.12.5/src/syscall/dll_windows.go:311 +0x42
syscall.(*LazyProc).Addr(0x11c459e0, 0xdda500)
	/home/jenkins/go1.12.5/src/syscall/dll_windows.go:318 +0x21
internal/syscall/windows.GetFileInformationByHandleEx(0x778, 0x9, 0x11c75a04, 0x8, 0x3, 0x11c1d348)
	/home/jenkins/go1.12.5/src/internal/syscall/windows/zsyscall_windows.go:353 +0x27
os.newFileStatFromGetFileInformationByHandle(0x11c908c0, 0x64, 0x778, 0x0, 0x11c908c0, 0x64)
	/home/jenkins/go1.12.5/src/os/types_windows.go:52 +0x135
os.(*File).Stat(0x11c42ed8, 0x984490, 0x11c42ed8, 0x0, 0x11c42ed8)
	/home/jenkins/go1.12.5/src/os/stat_windows.go:55 +0x133
io/ioutil.ReadFile(0x11c908c0, 0x64, 0x0, 0x0, 0x0, 0x0, 0x0)
	/home/jenkins/go1.12.5/src/io/ioutil/ioutil.go:62 +0xac
github.com/arduino/go-paths-helper.(*Path).ReadFile(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/go-paths-helper/paths.go:407
github.com/arduino/arduino-cli/arduino/cores/packageindex.LoadIndex(0x11c42ed0, 0xa29e60, 0x11c51ec0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/arduino/cores/packageindex/index.go:205 +0x2f
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).LoadPackageIndexFromFile(0x11d04ca0, 0x11c42ed0, 0x0, 0x0, 0x11c42ed0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/arduino/cores/packagemanager/package_manager.go:202 +0x25
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).loadPlatforms(0x11d04ca0, 0x11c9b5f0, 0x11c42e10, 0x7, 0x11c4cac8)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/arduino/cores/packagemanager/loader.go:186 +0xf4d
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).LoadHardwareFromDirectory(0x11d04ca0, 0x11c42bf8, 0xdda500, 0x703815)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/arduino/cores/packagemanager/loader.go:116 +0x43f
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).LoadHardwareFromDirectories(0x11d04ca0, 0x11c42c00, 0x2, 0x2, 0x1, 0x1)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/arduino/cores/packagemanager/loader.go:52 +0x4f
github.com/arduino/arduino-cli/legacy/builder.(*HardwareLoader).Run(0xdebc68, 0x11ce45a0, 0xdebc68, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/hardware_loader.go:43 +0xe4
github.com/arduino/arduino-cli/legacy/builder.(*ContainerSetupHardwareToolsLibsSketchAndProps).Run(0xdebc68, 0x11ce45a0, 0xdebc68, 0xdda500)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_setup.go:60 +0x266
github.com/arduino/arduino-cli/legacy/builder.runCommands(0x11ce45a0, 0x11c75f0c, 0x2, 0x2, 0xffffff01, 0x0, 0x4)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xbd
github.com/arduino/arduino-cli/legacy/builder.(*ParseHardwareAndDumpBuildProperties).Run(0x11c75f40, 0x11ce45a0, 0xdebbb4, 0x11c42c90)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:198 +0xbc
github.com/arduino/arduino-cli/legacy/builder.RunParseHardwareAndDumpBuildProperties(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:231
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:388 +0x772
C:\Documents and Settings\abhis cyber den\Desktop\arduino-1.8.10\arduino-builder returned 2

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

aniketstark, 2019-10-09:
i am in cyber dude i dont have pc thats why i making ino to bin

closed 2019-10-09 by BlueArduino20