adamff-dev/ESP8266-Captive-Portal #9: Changing SSID name

merdeux50, 2020-02-27:
Hello,
First I would like to say, holy shit, this is pretty cool!

Second I can only flash the .BIN file with NodeMCU, the Arduino source keeps giving me errors.

Arduino: 1.8.12 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

WiFi_Captive_Portal:31:3: error: expected ',' or ';' before 'String'
 } String arg=webServer.arg(argName);
   ^
WiFi_Captive_Portal:32:3: error: 'a' does not name a type
   a.replace("<","&lt;");a.replace(">","&gt;");
   ^
WiFi_Captive_Portal:32:25: error: 'a' does not name a type
   a.replace("<","&lt;");a.replace(">","&gt;");
                         ^
WiFi_Captive_Portal:33:3: error: 'a' does not name a type
   a.substring(0,200); return a; }
   ^
WiFi_Captive_Portal:33:23: error: expected unqualified-id before 'return'
   a.substring(0,200); return a; }
                       ^
WiFi_Captive_Portal:33:33: error: expected declaration before '}' token
   a.substring(0,200); return a; }
                                 ^
exit status 1
expected ',' or ';' before 'String'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

All I'm trying to do is change the SSID name from "My SSID" to something else, I don't touch any of the code.

Any help or pointers, something to read on how to fix this would be amazing.

Thank you.

234v244, 2020-03-21:
because he is hiding something

merdeux50, 2020-03-21:
What could it be? Murder??

guga330, 2020-03-29:
Any Luck? Im trying to run the NodeMCU on my mac... I still have a problem with it recognizing the NodeMCU. Perhaps you could give me some light ?

125K, 2020-03-29:
You have to follow the installation instructions:
https://github.com/125K/ESP8266_WiFi_Captive_Portal_2.0#installation-arduino-ide

closed 2020-03-29 by 125K