adamff-dev/ESP8266-Captive-Portal #26: Suggestion to add.

Impromenot, 2020-08-26:
Can i ask something, this supports deauth like fluxion or perform like fluxion just an captive portal?

I didn't try it but later. I dont know how to write code either.

If this is only an captive portal so i need 2 more devices to perform deauth & captive portal?

Of you could add something to your code like deauth it would be awesome too. (still not install) 

I would like to hear your answer, thanks.

125K, 2020-08-27:
This project is a captive portal, not a deauther. The ESP8266 is not able to deauth an AP while it's acting as an AP at the same time.

Alpha-1729, 2020-08-27:
There is a way
first deauth the other device for 1sec.
then make a ap for 5sec.
Is it possible.

Impromenot, 2020-08-27:
ohhh.. thanks so i need 2 device to performe a perfect attack. i already order 1 more to flash an deauther.

Alpha-1729, 2020-08-28:
I will inform you when I build it.
Contact me
https://alpha-1729.github.io/Alpha-1729/index.html

125K, 2020-08-28:
@Impromenot You better use two ESP8266 (as they are extremely cheap and the process would be more effective). The method that @Alpha-1729 suggests wouldn't work.

ShahriarShafin, 2020-09-08:
@125K can you please separate the html and css in other folder that we can modified it easily. Please help me to do that. I badly need it.

closed 2020-09-08 by 125K