adamff-dev/ESP8266-Captive-Portal #27: I Have A Big Request

Faisal-076, 2020-08-27:
Sir, I Appreciate and love your project
As you have created a great project
I believe about 80% people use TP Link Router as their own Wi-Fi Device
And this time i just want to say please make a wifi captive portal of TP Link Router
I have searched a lot for the TP Link Captive Portal
But i have just founded an website that provides me such information of TP Link Captive Portal
And also founded a github project repository of the source code
But in that website they mentioned about Micro Python and i really a beginner about that thing
So please sir, make a TP Link Captive Portal by modifying the source Code
Please translate the website if you're unable to understand
[Project Website](https://medium.com/@maaami98/evil-twin-attack-for-esp8266-3abca8ae0402) 
[Github Source Code](https://github.com/maaami98/Evil-Twin-For-ESP8266)

Alpha-1729, 2020-08-27:
You can design the captive portal with html and css  for the TP link captive portal.

Faisal-076, 2020-08-27:
@Alpha-1729 Brother i really don't know how to do that
Please make it for me
Requesting you

Alpha-1729, 2020-08-27:
I dont how the captive portal of a tplink look like.?

125K, 2020-08-27:
@Faisal-076 If you want a new design, you have to learn HTML and CSS. Otherwise, this project isn't for you.

Alpha-1729, 2020-08-27:
> 
> 
> @Faisal-076 If you want a new design, you have to learn HTML and CSS. Otherwise, this project isn't for you.

@125K Is there any way to add html and css separately in this repo.
instead of addding it  in the script.

Faisal-076, 2020-08-27:
> Sir, I Appreciate and love your project
> As you have created a great project
> I believe about 80% people use TP Link Router as their own Wi-Fi Device
> And this time i just want to say please make a [...]

Brother i have mentioned here all of the information and website link
And there you can also view the tp link captive portal image about how it looks like
I also added the source code link
Please brother...make it for me and also ask the project creator for help if you needed
@125K brother also help me please

125K, 2020-08-27:
@Alpha-1729 No, I'm afraid that's not possible.

Faisal-076, 2020-08-27:
@125K 
@Alpha-1729 
Please brother make it for me
I have given all information about that
I Request you

closed 2020-08-27 by 125K

ShahriarShafin, 2020-09-13:
I did it! @Faisal-076 

[image]

Faisal-076, 2020-09-14:
Brother i need it

Faisal-076, 2020-09-14:
Hlw brother i really need it

Alpha-1729, 2020-09-14:
Nice, how did you add the image.

ShahriarShafin, 2020-09-14:
@Alpha-1729 
1.goto https://www.base64-image.de/
2.convert your image to BASE64
3.and paste the file,   img src=HERE .

ShahriarShafin, 2020-09-14:
@Faisal-076 check your mail.

AnonDXT-XcX31, 2020-09-17:
> @Faisal-076 check your mail.

can i have the code too ????  please 
....
anonxcx31@gmail.com send it here brother