adamff-dev/ESP8266-Captive-Portal #38: exit status 1- Compilation error: no return statement in function returning non-void [-Werror=return-type]

angelomarzolla, 2023-01-31:
C:\Users\angel\Documents\Arduino\WiFi_Captive_Portal\WiFi_Captive_Portal.ino: In function 'String postedSSID()':
C:\Users\angel\Documents\Arduino\WiFi_Captive_Portal\WiFi_Captive_Portal.ino:108:1: error: no return statement in function returning non-void [-Werror=return-type]
  108 | }
      | ^
cc1plus.exe: some warnings being treated as errors

exit status 1

Compilation error: no return statement in function returning non-void [-Werror=return-type]

angelomarzolla, 2023-01-31:
.

Poko-Apps, 2023-02-10:
Consider visiting this until dev push some fix
https://github.com/adamff1/ESP8266-Captive-Portal/issues/39#issuecomment-1425159670

Danysevian, 2023-02-22:
> exit status 1- Compilation error: no return statement in function returning non-void [-Werror=return-type]

you just need add in the String postedSSID() this line:

return postedSSID;

should be like this:

String postedSSID()
{
  String postedSSID = input("s");
  newSSID = "<li><b>" + postedSSID + "</b></li>";
  for (int i = 0; i < postedSSID.length(); ++i)
  {
    EEPROM.write(i, postedSSID[i]);
  }
  EEPROM.write(postedSSID.length(), '\0');
  EEPROM.commit();
  WiFi.softAP(postedSSID);
  return postedSSID;                   //this is the adding line
}